ricott/sma.modbus #1: Can't add device since auto-discovery

TonD83, 2019-11-25:
Since auto-discovery of SMA devices was introduced, I can't use this app anymore. I'm using my own fork now, because I already implemented some improvements before you took over ownership of this app. However, now that this app seems to be getting some new love, I'd love to use the 'official' version again instead of my own fork. I'm using a SMA Sunny Boy 2.5, working flawlessly with the old app but now all I get is 'no device found'. Any suggestions?

SynoUser-NL, 2019-11-25:
Hi,
I have the same problem. I can not add my SMA Sunny Boy 2.5 as a new device.
The SMA device was installed in my home last week so I have no prior experience, but I would love to see a solution.
Thanks very much!
Den

ricott, 2019-11-25:
Hi,

So we have a consistent problem, that's good at least :)

@TonD83, you seem to have experience installing the app via CLI. If you run it in debug mode using `homey app run` - do you see an error or any type of message when you try to add the inverter?

All SMA devices should respond to the UDP multicast request, allowing them to be found automatically. 
Can you make sure UDP multicast is not blocked in your network? If it is then auto-discovery won't work.
If you run the app in debugging and see this log entry `Sending discovery datagram`, but then you don't see `Found inverter at: <IP>` - then I'd say we have a network problem related to UDP multicast. The response in my network is instant, there is basically no delay.

Regards,
Richard

TonD83, 2019-11-25:
Sure, I can have a look tonight. Now that you mention UDP.... I think our inverter has separate settings for modbus for TCP and for UDP and only TCP is enabled. Will check this tonight.

ricott, 2019-11-25:
In Sunny Explorer, I also have separate modbus settings for TCP and UDP. Here I only have TCP enabled on port 502. The UDP multicast for discovery doesn't go through modbus, SMA calls it speedwire. It is described here, https://www.sma.de/fileadmin/content/global/Partner/Documents/sma_developer/SpeedwireDD-TI-en-10.pdf.

Speaking of ports, that is another potential source for failure. The auto-discovery only allows us to find the IP of the inverter, not which port you have configured. In-app settings of the app in Homey, you must change the port if you have configured modbus to use anything else than default 502.

TonD83, 2019-11-25:
Thanks, I didn't change the default port 502 and speedwire is enabled I believe... Will update when I have looked at the debug statements...

SynoUser-NL, 2019-11-25:
Hi, 

Thanks for your quick reply.

Yes, the Inverter does have separate settings for TCP and UDP. Speedwire is enabled, Wifi is disabled  so using a wired connection.
I tried all combinations, but nothing worked. Port numbers were always left to default (502) for both settings.
As far as I can tell my router (Asus RT-AC87U) does not block broadcast traffic, but not sure how to check..

Best, Den

TonD83, 2019-11-25:
Sending discovery datagram
Collecting device info, found 0 devices.

The second line comes in almost instantly.

SynoUser-NL, 2019-11-25:
Besides the crappy modem part (ahemm..), I agree completely.
Automating stuff is nice, but sometimes the nicest thing about it is if you can turn it off.. :)

Thanks! Den

ricott, 2019-11-25:
I will think about this a little if there is a way to test if UDP is allowed in the router and if not show the input field for IP. Guess both of you don't have an Energy Meter or Home Manager 2? Since it relies completely on speedwire (UDP multicast).

TonD83, 2019-11-26:
Correct. Weird thing is, I almost immediately get the 0 devices found message. I only have basic knowledge of multicastimg, so pardon my ignorance, but I would expect to see some kind of timeout maybe?

ricott, 2019-11-26:
If you download the latest code from this repo and install using CLI, then if auto-discovery doesn't find any inverters it should display the old manual way for you. Please try it out and let me know how it works for you.

TonD83, 2019-11-27:
Works like a charm, thanks for looking into this so quickly! 

```
2019-11-27 09:50:11 [log] [ManagerDrivers] [inverter] Showing settings for first time, lets do discovery
Sending discovery datagram
Collecting device info, found 0 devices.
2019-11-27 09:50:17 [log] [ManagerDrivers] [inverter] No inverters found using auto-discovery, fallback to manual entry
{ host: '192.168.1.14', port: 502, autoClose: true }
Client connected on IP '192.168.1.14'
{ host: '192.168.1.14', port: 502, autoClose: true }
Client connected on IP '192.168.1.14'
Client closed for IP '192.168.1.14'
2019-11-27 09:50:45 [log] [ManagerDrivers] [inverter] Adding to devices: SB 2.5-1VL-40
Client closed for IP '192.168.1.14'
2019-11-27 09:50:46 [log] [ManagerDrivers] [inverter] [0] SMA inverter initiated, 'SB 2.5-1VL-40'
{ host: '192.168.1.14', port: 502, refreshInterval: 5 }
Client connected on IP '192.168.1.14'
Timers initialized
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Setting up capabilities
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'measure_power'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'meter_power'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'measure_voltage'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'measure_yield'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'operational_status.health'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'operational_status'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Device has capability 'measure_voltage.dcA'
2019-11-27 09:50:47 [log] [ManagerDrivers] [inverter] [0] Removing capability 'measure_voltage.dcB'
```

ricott, 2019-11-27:
I cleaned it up a little more and added a first page explaining the process. It is available now as version 2.0.3 that is available as a test version here: https://homey.app/sv-se/app/sma.modbus/SMA-Solar/test/

Happy to get your feedback before I release it to prod.

SynoUser-NL, 2019-11-27:
Hi,

After installing the test version, adding the device was a breeze!
Thanks very much for your quick action!! Excellent work..

Best wishes, Den

TonD83, 2019-11-27:
Same here, readme update (even though I'm not sure why you pointed us to the Svenska language Homey App store :D) and wizard explanation are clear as well.

ricott, 2019-11-27:
Thanks for checking!

closed 2019-11-27 by ricott