ricott/sma.modbus #2: Device Condition instead of Operational Status

SynoUser-NL, 2019-12-01:
Hi,

Again, thanks very much for modifying the app so I can add my SMA Inverter by specifying the IP address!
When using the app in Homey I found that I can monitor the operational status of the device, but I can not make a flow that triggers when the device condition is changed from OK to something else..
I'm not sure I can achieve the same thing by monitoring the operational status (MPP means nothing to me.. :) ) but it would make it a lot more clear for me if I could monitor on device condition. 
Would you by any chance be able to add this to your excellent app?

Thanks very much!
DenW

ricott, 2019-12-03:
Hi,

Sure, that can be added. Will include it in the next release.
You can see possible status for both of the fields here, https://github.com/ricott/sma.modbus/blob/master/lib/decodeData.js. Look at decodeOperatingStatus and decodeConditionType.

/Richard

ricott, 2019-12-11:
Released in version 2.0.4

closed 2019-12-11 by ricott

SynoUser-NL, 2019-12-11:
Thanks very much for your quick enhancement!
Much appreciated.. :)