ricott/sma.modbus #3: Daily production

mvaneck83, 2019-12-03:
Ey, 
What happened with the daily production read out? I used it in a flow but it isn't available anymore. 

Thanks for a reply,
Martijn

ricott, 2019-12-03:
Hi,

When did you see it went missing, I guess in first v2 release? 
What is the device type in the advanced setting on your inverter? Since version 2 uses custom mapping per inverter type I can't look at this without knowing which device type it is. 

/Richard

mvaneck83, 2019-12-03:
EY Richard,
Thanks for quick reply. Last week I noticed it for the first time. So I think that it is since version 2 yes. I have a sma sunny boy 3.0-1av-40. In advanced settings I see “UNKOWN (9319)” as type and serial number “1992108605”. See photo attachment 
[image]

ricott, 2019-12-03:
Thanks, '9319' gave me everything I needed. Support for all capabilities will come in the next release. Traveling this week so releasing next week.

mvaneck83, 2019-12-04:
You’re the man!

kind regards,

Martijn van Eck

closed 2019-12-11 by mvaneck83