ricott/sma.modbus #6: Sunny Boy 2.5 1VL40

MichaelHendriks, 2020-04-04:
The Homey app can’t find my Sunny Boy 2.5 1VL40. Not even when I have entered the IP address of my Sunny Boy.
But if I want to log into my Sunny Portal I have to enter user data, I cannot enter this in the Homey SMA app?

Please help.

ricott, 2020-04-04:
It should be a supported inverter, please make sure you use the inverter device type and not storage.
Further, you need to enable Modbus on your inverter. This is described in the first post here, https://community.athom.com/t/app-sma-solar-devices/696, under section "Enable modbus in your inverter/storage product".

Reading values via Modbus doesn't require any authentication.

MichaelHendriks, 2020-04-04:
Via Sunny Explorer installer (password: 1111) settings I was able to enable (yes) “Modbus TCP”.
Thanks for your help!

ricott, 2020-04-04:
Great that it worked!

closed 2020-04-04 by ricott