ricott/sma.modbus #7: IP address

kwezel, 2020-09-07:
Dear, 
Many thanks for the work to make the app .

I try to install it, app succeeds, but then need to install in the inverter, he asks the IP address, I gef the correct address in as much as 172.xxx.
 My inverter is a sunny boy SB 4000TL-21.
I can see everything on the sunnyportal and app my android cell phone.

Do you have a solution for that, would love to work with your appa.

Thank you very much,
Walter

ricott, 2020-09-07:
Hi,

Have you enabled Modbus on your inverter? 
The procedure is described in the first post here, https://community.athom.com/t/app-sma-solar-devices/696, under section "Enable modbus in your inverter/storage product".

ricott, 2021-04-21:
I assume this problem was sorted out :)

closed 2021-04-21 by ricott