ricott/sma.modbus #9: Stop update the app

svarla, 2021-04-21:
The reporting from the energy meter stop reporting value to the app. If I connect direct I can see all the value 
Svenne

ricott, 2021-04-21:
The energy meter is usually pretty solid since it doesn't connect to any host. It opens a port locally on the Homey and accepts UDP datagrams that are sent via multicast from the energy meter / home manager.
If you click advanced settings on the energy device in Homey, do you see any error in the "most recent error" field?
Do you see any info like serial number or software version there?

Have you tried to restart the SMA Homey app, or your entire Homey? If something got "stuck" in a bad state.

svarla, 2021-04-21:
This is what it looks like
It functioning a short while after restart 
I have also tried to reinstalling the app

Svenne
Sven Larsson
Storholmens Allservice AB
+46 70600 55 15

jonorunde, 2021-05-02:
Hi, I have the same issue

A little more info;
* If I remove the device, I cannot add it again (no energy meters found) - but if I take power of and on, on the energy meter, I can add it, and it works for a little while.
* rebooting the Homey does not help
* I have a raspberry pi running with some python code, that gives me data from the energy meter, and that works flawlessly
* I will try to take power, so I can add it again - and see if I get any error messages in the system

ricott, 2021-05-04:
@jonorunde, would be great if you could install the app on your Homey using CLI and command `homey app run`. Then you get the full logging to your laptop's terminal/command window and we can see what goes wrong.

There is also a test file in the test folder called test_em_discovery.js. If you start it locally you can test the discovery logic from your laptop, ie it will be the Homey app code, but executed on your laptop. `node test/test_em_discovery.js`

If it works on your laptop only reason for it to not work on your Homey I guess is network config. But that would be confirmed if you also install the app via CLI.

jonorunde, 2021-05-06:
I have started the app on pc, and it is running OK  - should I see logging here, when I add a device on the homey app - or can I somehow add the device via cli ?
When I do "add device, it still shows "No Devices found"
This is all I get;

✓ Installing Homey App on `Homey Pro` (https://XXX
✓ Homey App `sma.modbus` successfully installed
✓ Running `sma.modbus`, press CTRL+C to quit
 — Profile your app's performance at https://go.athom.com/app-profiling?homey=XXX
─────────────── Logging stdout & stderr ───────────────
2021-05-06 12:47:25 [log] [SmaModbusApp] Initializing SMA Modbus app ...
2021-05-06 12:47:25 [log] [ManagerDrivers] [energy] SMA energy driver has been initialized
2021-05-06 12:47:25 [log] [ManagerDrivers] [energy] Registering flows
2021-05-06 12:47:25 [log] [ManagerDrivers] [energy] - flow 'trigger.phase_threshold_triggered'
2021-05-06 12:47:25 [log] [ManagerDrivers] [energy] - flow 'condition.phaseUtilized'
2021-05-06 12:47:25 [log] [ManagerDrivers] [energy] - flow 'condition.anyPhaseUtilized'
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] SMA Inverter driver has been initialized
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] Registering flows
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] - flow 'trigger.inverter_status_changed'
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] - flow 'trigger.inverter_condition_changed'
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] - flow 'condition.isInverterDailyYield'
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] - flow 'condition.isInverterStatus'
2021-05-06 12:47:25 [log] [ManagerDrivers] [inverter] - flow 'condition.power_condition'
2021-05-06 12:47:25 [log] [ManagerDrivers] [pvoutput] SMA PVOutput driver has been initialized
2021-05-06 12:47:25 [log] [ManagerDrivers] [pvoutput] Registering flows
2021-05-06 12:47:25 [log] [ManagerDrivers] [summary] SMA summary driver has been initialized

Then I add device in app, and no logging occurs

When I try to run the test;
node test\test_em_discovery.js
[]

And that is it..

From Pi, using smaem;
pi@raspberrypi:~/smaem $ ./sma-em-measurement.py

SMA-EM Serial:3009869809
----sum----
P: consume:6255.8W 5389.513kWh supply:0.0W 420.5974kWh
S: consume:6338.3VA 5523.4134kVAh supply:0.0VA 454.3365VAh
Q: cap 0.0var 7.7701kvarh ind 1019.1var 1010.1017kvarh
cos phi:0.987°
----L1----
P: consume:3392.2W 2816.8112kWh supply:0.0W 199.9261kWh
S: consume:3649.6VA 3596.9198kVAh supply:0.0VA 291.6389kVAh
Q: cap 0.0var 0.017kvarh ind 1346.2var 2145.088kvarh
U: 240.234V I:15.248A cos phi:0.929°
----L2----
P: consume:2863.6W 2713.5876kWh supply:0.0W 361.557kWh
S: consume:2882.2VA 3061.2706kVAh supply:0.0VA 429.3747kVAh
Q: cap 327.1var 1191.4467kvarh ind 0.0var 48.7071kvarh
U: 241.234V I:12.017A cos phi:0.994°
----L3----
P: consume:0.0W 0.0kWh supply:0.0W 0.0kWh
S: consume:0.0VA 0.0kVAh supply:0.0VA 0.0kVAh
Q: cap 0.0var 0.0kvarh ind 0.0var 0.0kvarh
U: 0.0V I:0.0A cos phi:0.0°
Version: 2.6.6.R|020606

ricott, 2021-05-06:
Interesting, so the test discovery doesn't find your Energy Meter. I'm on the same firmware version - and it finds mine.
Can you link to the python script for me to understand what it does?

jonorunde, 2021-05-06:
https://github.com/datenschuft/SMA-EM
I use this project to get data to the Pi

ricott, 2021-05-06:
The timer only waits for 5 seconds to "discover" an energy meter. My home manager sends datagrams once per second, don't know if this is configurable? In the test_em_discovery.js file there is a sleep for 5000 ms, if you change that to 60000 for instance - does it "discover" the energy meter then? Or you see using the python script that you gets updates once per second?

Didn't spot anything immediately looking at the python script, ie difference. Same multicast address and port that I use. But will spend some more time this evening.

jonorunde, 2021-05-06:
Changed timeout, and I get the same result on the test

jonorunde, 2021-05-06:
I am not binding to a spesific IP no - and the reason I do not have L#, is that I have IT-net, no not "proper" 3-phase, but the EM is handling that fine :)

ricott, 2021-05-06:
If you refresh files from github I made a minor change to two files so if you run test_em_discovery.js again it will print the raw datagrams. If you don't get any printouts now I have to think a little more about what to do next

ricott, 2021-05-06:
The code is hardcoded to use IPv4, by using `udp4`.
Are you using IPv4 or IPv6?

jonorunde, 2021-05-06:
Am using ipv4, pretty standard network
Did not get any more output from the new version...

jonorunde, 2021-05-06:
I am not running on the pi - but I probably could - but, now it actually works on my windows PC ;

node test\test_em_discovery.js
Listen via UDPv4 on Device 'Ethernet' with IP '192.168.88.189' on Port '9522' for Multicast IP '239.12.255.254'
Listen via UDPv4 on Device 'Npcap Loopback Adapter' with IP '169.254.193.111' on Port '9522' for Multicast IP '239.12.255.254'
Message:  534d4100000402a000000001024c001060690174b366f7f143b49b1f000104000000386700010800000000048b686818000204000000000000020800000000005ae8e9000003040000000000000308000000000001af78b80004040000001b120004080000000000da44d8080009040000003e900009080000000004a85e6a30000a040000000000000a080000000000622f28a0000d040000000386000e04000000c3060015040000001b16001508000000000260f7d098001604000000000000160800000000002aeafe28001704000000000000170800000000000000ef100018040000001a740018080000000001ce96cba0001d0400000025dc001d08000000000309077a90001e040000000000001e0800000000003ea5eb30001f040000001080002004000003b38a00210400000002cb0029040000001d520029080000000002494e11a0002a040000000000002a0800000000004edb6390002b040000000000002b0800000000010079bae8002c04000000009e002c0800000000000a793ad80031040000001d53003108000000000293f8aa30003204000000000000320800000000005d8047f80033040000000c1e003404000003b93c00350400000003e8003d040000000000003d08000000000000000000003e040000000000003e08000000000000000000003f040000000000003f08000000000000000000004004000000000000400800000000000000000000450400000000000045080000000000000000000046040000000000004608000000000000000000004704000000000000480400000000000049040000000000900000000206065200000000
Adding device: 3009869809
Message:  534d4100000402a000000001024c001060690174b366f7f143b49f08000104000000380c00010800000000048b686db8000204000000000000020800000000005ae8e9000003040000000000000308000000000001af78b80004040000001af50004080000000000da44dad80009040000003e310009080000000004a85e6fd0000a040000000000000a080000000000622f28a0000d040000000385000e04000000c3020015040000001b16001508000000000260f7d368001604000000000000160800000000002aeafe28001704000000000000170800000000000000ef100018040000001a760018080000000001ce96ce70001d0400000025de001d08000000000309077ec8001e040000000000001e0800000000003ea5eb30001f040000001080002004000003b3d300210400000002cb0029040000001cf60029080000000002494e15d8002a040000000000002a0800000000004edb6390002b040000000000002b0800000000010079bae8002c040000000080002c0800000000000a793ad80031040000001cf7003108000000000293f8ad00003204000000000000320800000000005d8047f80033040000000bf9003404000003b8f000350400000003e8003d040000000000003d08000000000000000000003e040000000000003e08000000000000000000003f040000000000003f08000000000000000000004004000000000000400800000000000000000000450400000000000045080000000000000000000046040000000000004608000000000000000000004704000000000000480400000000000049040000000000900000000206065200000000
Message:  534d4100000402a000000001024c001060690174b366f7f143b4a2f0000104000000387a00010800000000048b687358000204000000000000020800000000005ae8e9000003040000000000000308000000000001af78b80004040000001ad70004080000000000da44dda80009040000003e880009080000000004a85e76d8000a040000000000000a080000000000622f28a0000d040000000387000e04000000c3020015040000001b7c001508000000000260f7d638001604000000000000160800000000002aeafe28001704000000000000170800000000000000ef100018040000001a590018080000000001ce96d140001d040000002613001d08000000000309078198001e040000000000001e0800000000003ea5eb30001f040000001099002004000003b37d00210400000002d20029040000001cfe0029080000000002494e18a8002a040000000000002a0800000000004edb6390002b040000000000002b0800000000010079bae8002c04000000007e002c0800000000000a793ad80031040000001cff003108000000000293f8afd0003204000000000000320800000000005d8047f80033040000000bfd003404000003b8e300350400000003e8003d040000000000003d08000000000000000000003e040000000000003e08000000000000000000003f040000000000003f08000000000000000000004004000000000000400800000000000000000000450400000000000045080000000000000000000046040000000000004608000000000000000000004704000000000000480400000000000049040000000000900000000206065200000000
Message:  534d4100000402a000000001024c001060690174b366f7f143b4a6d7000104000000383300010800000000048b6878f8000204000000000000020800000000005ae8e9000003040000000000000308000000000001af78b80004040000001af80004080000000000da44e0780009040000003e560009080000000004a85e7c78000a040000000000000a080000000000622f28a0000d040000000386000e04000000c2f90015040000001b25001508000000000260f7d908001604000000000000160800000000002aeafe28001704000000000000170800000000000000ef100018040000001a810018080000000001ce96d410001d0400000025f1001d080000000003090785d0001e040000000000001e0800000000003ea5eb30001f040000001087002004000003b3fd00210400000002cc0029040000001d0e0029080000000002494e1b78002a040000000000002a0800000000004edb6390002b040000000000002b0800000000010079bae8002c040000000076002c0800000000000a793c400031040000001d0f003108000000000293f8b2a0003204000000000000320800000000005d8047f80033040000000c02003404000003b97300350400000003e8003d040000000000003d08000000000000000000003e040000000000003e08000000000000000000003f040000000000003f08000000000000000000004004000000000000400800000000000000000000450400000000000045080000000000000000000046040000000000004608000000000000000000004704000000000000480400000000000049040000000000900000000206065200000000
[... cut ...] 570003204000000000000320800000000005d8047f80033040000000c06003404000003b92600350400000003e8003d040000000000003d08000000000000000000003e040000000000003e08000000000000000000003f040000000000003f08000000000000000000004004000000000000400800000000000000000000450400000000000045080000000000000000000046040000000000004608000000000000000000004704000000000000480400000000000049040000000000900000000206065200000000

[ { name: 'Energy Meter (3009869809)', data: { id: 3009869809 } } ]

So that looks promising :)

ricott, 2021-05-07:
Lets see, I'll issue a new version

ricott, 2021-05-07:
The change is available as a test version, can you try it out
https://homey.app/a/sma.modbus/test/

jonorunde, 2021-05-07:
I tried the test-version - when I try to add the EM, I get the "experimental"-text in the app, so looks correct, but unfortunately it still does not see the EM..

ricott, 2021-05-07:
There was a bug that if you already had an energy meter added you couldn't search for new ones - had to set a flag to reuse the socket. Version 2.2.2 is now available as test version.
If you install via CLI, homey app run, then you should see this line. Of course with the IP of your Homey.
`Listen via UDPv4 on Device 'wlan0' with IP '192.168.200.215' on Port '9522' for Multicast IP '239.12.255.254'`

If this doesn't work I suspect there are some network differences between your laptop and the Pi that works vs your Homey that doesn't work. Ie the Homey doesn't receive the udp multicast traffic.

ricott, 2021-05-21:
@jonorunde , and progress on this?

jonorunde, 2021-05-27:
Sorry, I have been away for some time, back home now :)
I have installed 2.2.2 - app on homey is correct version, but still I get "no sma energy meters found...

ricott, 2021-11-07:
Got a message some time ago from @svarla that it started to work for him after changing the wifi equipment. I can't see that this can be anything else than a network config issue.

closed 2021-11-07 by ricott