ricott/sma.modbus #13: Grid Power

Michael-1880, 2021-11-05:
Dear Richard, first of all. Thanks for making SMA part of Homey!
Maybe there is a small problem.

In the Energy Meter, the 'Grid Power' and also 'L1 power' is showing the power (as it should) but then every few seconds it drops to zero. This is annoying in a flow where you want to make an action when the grid power is on.

If anything have to with it. I use it with a Sunny Island 6.0 v13 in combination with a generator.
So when the generator is on, I want to disable al high peak demanding devices.

If I can be of any help, please let me know.

Best regards

ricott, 2021-11-07:
Hi!

You are using an SMA Home Manager 2, ie the Energy Meter device type as I understand it?
If you are a bit tech-savvy you can install nodejs on your laptop and run one of the test files that are available here on github. The file https://github.com/ricott/sma.modbus/blob/master/test/test_em.js you can start using `node test_em.js` on your laptop. This will show the result from your Home Manager using the same code that is used on your homey. You should be able to observe if your Home Manager sends 0 values as you observe in Homey.

I realize now that you need to create a file called config.js in the test folder. This file should look like below, where the serial number must be the serial number of your Home Manager. The serial number you find in advanced settings on your device in Homey or you can run the other file in the test folder called `test_em_discovery.js` which will discover sma home manager devices on your network and print serial numbers found.

```
var config = {
    hm2: {
        serial: <insert your serial here>
    }
};
module.exports = config;
```

Michael-1880, 2021-11-09:
Hello,

Thanks for your feeback! I'm not really a tech-savvy but I tried your
suggestion.
After several hours I managed this :-)

[image: image.png]
The discovery node is adding a device with an unknown serial number in my
SMA system.
The serial nummers I know are these:

SMA Home Manager 2:                3009887355       (192.168.1.107)

SMA Energy Meter:                       1900241434      (192.168.1.131)

SMA Sunny Island:                         3007957274      (192.168.1.103)

SMA Sunny Boy 4.0:                      3007638934      (192.168.1.106)

If I use enter on of the above serial number, or also any number and run
the test_em node I get this:
[image: image.png]
The 192.168.1.109 is the IP of my laptop...

I'll try sniffing eround in this code but this is a step to far for me I
thinks :-)

Maybe the problem is that I use 2 Energy Meters?
The one integrated in my Home Manager 2 and an Energy meter witch is only
on when I run my generator...
When the generator is off, then the Home Manager gives the correct Watts.
But the node test_em still gives the above screenshot...

What I was looking for in Homey is just the function when the generator is
on, shut down al my pumps.
The pumps aks for a very high peak of power witch shuts down everything...

Again, already many thanks for your feedback and teaching me this 'node'
thing :-)

Thanks!!

Best Regards
Michael.

PS: I work at a wholesaler of SMA stuff... If I every can help you!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ricott, 2021-11-19:
Think I may have found the reason for the 0 values.
There is a test version out which you can try, https://homey.app/a/sma.modbus/test/

Michael-1880, 2021-11-24:
Hi, it seems to work now!
Thanks a lot.

I work as the responsable of a group of wholesalers in ventilation and
solar systems (i.e. SMA).
If I can do something for you, let me know.

Best Regards

Michael Heylen
+32 493 44 84 43

closed 2021-11-24 by ricott