ricott/sma.modbus #14: connecting HM2.0

KevinHomey, 2021-11-16:
Hi,

Thanks for the plugin.
For me it is not finding all devices.
A SB3.6 is always found after enabling TCP 502. 
The SB5000LT21 was found and connected, but I removed all my devices and uninstalled the SMA plugin and restarted the install procedure. 
I did this because the the Homemangager 2.0 is not to be found by the SMA plugin.

How can I add the HM2.0? I added the IP address of the HM2.0 in modbus configuration of the HM2.0, ID 1, MOXA1214 profile..

Kind regards,
Kevin

ricott, 2021-11-19:
Is your Home Manager connected to your switch or directly to your Inverter?
My understanding is that the HM is only broadcasting its UDP multicast datagrams on the network if connected to a switch

KevinHomey, 2021-11-23:
Hi,
All devices are on the same router/switch. 
When adding an energy meter I get error; Cannot read property 'name' of indefined.

Thx Kevin

ricott, 2021-11-23:
There is a new test version since a few days, can you try using it?
https://homey.app/a/sma.modbus/test/

KevinHomey, 2021-11-23:
Hi Ricott,
Same issue on 2.3.2
I also have a second invertor that needs to be found.
TCP is active on port 502, but error = No new devices found.
For the HM2.0, same error on this version.
Do I need to setup Modbus on the HM2.0?
Would it help to reset the homey completely and rebuild from new? I don't have anything special setup in the homey yet, so no problem.

Thx
Kevin

ricott, 2021-11-24:
As for your inverter, can you connect to port 502 from a laptop on the same network as your Homey is connected to?
On a mac you can type `nc -vz <inverter ip> 502` (full example: `nc -vz 192.168.200.249 502`) and if it shows success you have connectivity from the network and your Homey should be able to connect.
From Windows, I guess you need telnet installed.
If this works but the SMA app can't auto-detect the inverter, then you have something in your network that blocks UDP multicast (not uncommon if you use for instance your ISP router they provide you with). Then you simply use the window that is displayed to manually add the IP of your inverter - this works even if UDP multicast is blocked in your network.

As for HM2, please try [this](https://github.com/ricott/sma.modbus/issues/13#issuecomment-962624393). 
Are you saying that auto-detect detects an HM2 but when you click next to add it you get an error that says "Cannot read property 'name' of undefined."?

KevinHomey, 2022-02-03:
Hi Ricott,
Still no access to my 2nd invertor or HM2 with latest updates.
Also no popup to enter IP address manually.
Any advice??
Thx
Kevin

ricott, 2022-02-03:
My experience so far is that if HM2 is not found there is a network issue.
HM2 issues UDP datagrams using multicast into the local network. The SMA app simply opens a port locally on the Homey Pro and listens to these UDP datagrams. It doesn't connect to the HM2 in any way.
* The HM2 needs to be connected to your network, and not directly to your Inverter.
* You need to make sure that you have UDP multicast enabled between your Homey Pro and your HM2.

>When I try to to connect with homey, 'No new devices have been found'. No NEXT to enter the IP manually.
There could be something that I can look into, let me check. If the app finds an interverter it won't show the add manually view. In this case, I guess it finds the inverter you have already added.

ricott, 2022-02-04:
With this [test version](https://homey.app/a/sma.modbus/test/), you should see the manual add with IP view when adding an inverter.

The only time the auto-discovery doesn't work is for the same reason as HM2 doesn't work. 
>You need to make sure that you have UDP multicast enabled between your Homey Pro and your HM2.

But if TCP Modbus is enabled on the inverter then it will work to add it manually, for inverters it is only auto-discovery that relies on UDP multicast - normal operation use TCP to connect to the IP of the inverter.

KevinHomey, 2022-03-26:
Hi
I tested the test version but no help on that. 
Now tried latest version. Problem stays the same. 
My older 5000lt21 is not found. 
I know the correct ip from the screen on the device and port 502 tcp is open. 
Also Homemanager data is not found. 
Can you be of any assistance?
We can do a login on my devices if you want to troubleshoot with my setup. 
I now also have a Sma battery boy 5.0. That is connected to homey and works fine. 

Thx K≡vin

Mark323513, 2023-05-12:
I have updated to Homey Pro 2023

With this update occurs some issues
The Energy Manager can't be integrated 

The App trys to find new meters, but can't find new ones
"cannot read properties of null (reading "length")"

ricott, 2023-05-12:
This particular problem I see I introduced in the last version, there is a new test version that fixes that bug

Mark323513, 2023-05-12:
Hi Richard,I tested this with the last test version at Homey App Shop/Test Von meinem iPad gesendet

Mark323513, 2023-05-12:
With the new version I deleted the energy manager and tried to add it again.This doesn’t work, Homey can’t find the Energy meterVon meinem iPad gesendet

ricott, 2023-05-13:
The fix was aimed at removing the error `cannot read properties of null (reading "length")`
As stated above, all problems related to not being able to add the HM2 are around network connectivity. Segmented networks, not allowing UDP between networks where HM is vs Homey. Or optionally UDP multicast is turned off in the network in general - this is often the default setting if you use a router from your ISP.

> HM2 issues UDP datagrams using multicast into the local network. The SMA app simply opens a port locally on the Homey Pro and listens to these UDP datagrams. It doesn't connect to the HM2 in any way.

> The HM2 needs to be connected to your network, and not directly to your Inverter.
You need to make sure that you have UDP multicast enabled between your Homey Pro and your HM2.

Mark323513, 2023-05-13:
Thanks Richard,Will try to change the switches. Had a similar problem with a hue bridge.In my current network setup a Netgear router in AP Modus is linked to a Fritz.box Cable LAN. The HM2 linked directly at the Fritz.Box.The  Homey 2023 is linked via LAN cable to the Netgear Router.This might block the UDP communication.Will change the LAN Setup and hope this solves the problem CheersMartinVon meinem iPad gesendet

Mark323513, 2023-05-13:
Hello Richard,Change the Internet connection has helped.Now the Homey is connected to a client Fritz.Box and integrated in the Fritz Mesh.SMA converters we’re foundCheersMartinVon meinem iPad gesendet

closed 2023-05-16 by ricott