ricott/sma.modbus #17: Try to Connect the Sunny boy 3.0

monodie, 2022-01-13:
I try to Connect A Sunny Boy 3.0 (SB3.0-1AV-41), a inverter without  battery, and to do so I choose the Inverter in the Homey SMA app. The app recognise the unit not, and present the box to enter the IP-adress. After that, the installation rails, “wrong op or Port” it’s says.
The Port is correct, 502, and the IP also of course, see the images added. Do you know this model is supported?
[image]
[image]
[image]

monodie, 2022-01-15:
Thank you for your quick reply! I was so hard focusing on the portnumber, completly blind for the status "nee"", and off course, than it will never works. It is not that smart! I have make a request to the installer, because thge "user": account has not the credentials to change that part, that can do by the "installer" account, I do not have the password of that account. I think it wil works after change the value in "ja" (yes). Thanks for your support.

closed 2022-01-15 by ricott

nielslbeck, 2022-05-13:
I've been trying to get the Homey app to work multiple times but every time I've received the message that my SunnyBoy inverter couldn't be found. Every time I've looked at the configuration and concluded that the SB was set to port 502 as expected.

Apparently I've never looked at the setting right above the port number setting and noticed that communication was turned off.

Thank you for providing the answer that of course the communication should be activated on the SunnyBoy. Fortunately I got the installer password, so I could change the setting my self. Now the app is running like a charm on my Homey. Thank you for a great app, @ricott 👍