ricott/sma.modbus #19: Some data to PVOutput missing in upload, (Since last update?)

Not2Fast, 2022-02-02:
Hi Robert,

Seems that since last update, upload to PVOutput.org is not complete anymore. 
Power is updated, but Energy, Efficiency, etc. is missing.
See screenshots:

[image]

[image]

[image]

Any idea?
Thanks!

ricott, 2022-02-02:
I can observe the same thing myself, will have a look

[image]

ricott, 2022-02-02:
Just issued v2.3.4
A unit conversion problem, sent yield as kWh and not as Wh.

closed 2022-02-02 by ricott

Not2Fast, 2022-02-02:
Confirmed!

[image]

Thanks for the quick support Robert!