ricott/sma.modbus #20: Homey SMA Sunny Tripower 20000TL-30

kgbengt, 2022-03-05:
When i try to add my inverter it does not find it, SMA Sunny Tripower 20000TL-30. Is it not supported?
I have also entered the ip adress of the inverter.
Homey early 2019, latest firmware.
SMA Energy app version 2.3.6

ricott, 2022-03-06:
It is supported, I myself have an STP 25000TL-30
If the inverter is not found using auto-discovery there can be two reasons
- Your network is not supporting UDP multicast
- You have not enabled Modbus on port 502. Modbus is by default not enabled, you have to enable it manually for the app to work. Guide here: https://community.homey.app/t/app-pro-sma-solar-devices/696#enable-modbus-in-your-inverterstorage-product-9

kgbengt, 2022-03-06:
Ok, thanks for the answer. Looks like my installer changed the code in
Sunny Explorer, I will try to contact them (Ecokraft) and see if i can get
it.
Is it the same procedure with the Home Manager 2? I will soon get one of
those installed, and I suppose I don't need Homey to talk to both?

ricott, 2022-03-06:
They usually change the installer code, but I got Svea Solar to tell me the code. Maybe more by accident since they probably used the same code on all inverters they install.

For HM2 it is easier, but you have to have UDP multicast support in your network for that to work at all. Which you normally have unless you have segmented your network and potentially block traffic between networks - or you use the router from your ISP and they block UDP multicast in that equipment.

By default, the HM2 sends out UDP multicast datagrams into your network, once per second. The only thing the Homey app does is capture and use the info. So there is no change needed in the HM2 config.

kgbengt, 2022-03-06:
Ok, sounds good! I hope Ecokraft can send me the code also, or log in and
change the setting for me. Are there more interesting settings to play with
on the inverter?
I've got a Dream Machine Pro and there is an option to enable/disable
multicast, so this shouldn't be a problem.

closed 2022-04-26 by ricott