ricott/sma.modbus #21: Connection error

Bvb26, 2022-04-16:
Hi - I’ve used the sma energy app on my Homey for the last two years without any issues. Recently I’m getting different error messages. When I restart the app it loads the data, later it seems frozen. Latest error message is:

2022-04-16T17:03:17.337Z
Error: Failed to read '30783' for device type 'STP 7000TL-20'
    at modbusReading (/lib/sma.js:221:24)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)

any suggestions?

ricott, 2022-04-26:
Do you still see this problem?
I haven't seen any other complaints about devices not getting updates since v2.3.8.

Bvb26, 2022-04-26:
Hi Richard - thanks for getting back. I still have the same problem. When I restart the app on Homey it will make the connection correctly, but after a while it loses connection, intermittently as it sometimes works. The latest errormessage I got was:

2022-04-26T20:04:41.375Z
Error: connect ETIMEDOUT 192.168.1.142:502
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1137:16)

Bvb26, 2022-04-27:
Removed sma energy app today, restarted Homey, reinstalled sma app and added the inverter. Still the same error - initial connection ok but after a while it times out.

2022-04-27T11:20:54.931Z
Error: connect ETIMEDOUT 192.168.1.142:502
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1137:16)

Bvb26, 2022-04-28:
Problem solved. Did some further research yesterday and tried to read data with a modbus reader. This gave the same intermittent connection issues, pinpointing the problem to the inverter. Reloaded inverter firmware and reset the inverter. problem solved.