ricott/sma.modbus #22: SMA Sunny Tripower 8.0 only show L1

Ch1msie, 2022-04-25:
Added today the SMA Sunny Tripower 8.0 inverter to homey. This is a 3 phase inverter, unfortunately Homey only shows 1 Phase, the L1. Is this a know bug?

[image]

Ch1msie, 2022-04-26:
[image]
[image]

ricott, 2022-04-26:
Mainly interested in the device type property, but I guess it matches the name of your device, ie `STP 8.0-3AV-40`.
This inverter is for sure supported and only missing daily yield, which the app instead calculates manually for you.
I misunderstood your initial question though, the app only shows voltage for L1, even for 3-phase inverters. To limit the number of capabilities.

If you have an SMA Home Manager 2 it will show you the voltage of all three phases.

ricott, 2022-08-12:
Got this request also from another user so decided to add L2 and L3. 
There is a test version 2.3.9, please try it out.

closed 2022-11-10 by ricott