ricott/sma.modbus #23: Energy Meter data gets stuck

SWarnberg, 2023-04-01:
Hi

I've connected an Energy Meter 2.0 to the app, and initially the data comes in fine. However, after some minutes the data gets stuck and the same values is shown forever in the Homey app. See picture below.

[image]

After a restart of the app, it takes 7 minutes before it gets stuck again:

[image]

Any idea of the cause and solution?
I'll send a diagnostics report as well from the app.

Staffan

SWarnberg, 2023-04-09:
Hi @ricott 

Some more details. I ran the code and find that the messages stops coming in on the client. This happens every time after 4m20s. There is no 'error' or 'close' message coming in either.

```
function initListenersAndConnect(self) {

    self.client = dgram.createSocket({ type: 'udp4' }); //, reuseAddr: true });

    self.client.on("message", function (message, rinfo) {
```

SWarnberg, 2023-04-10:
I added a PR #24, which is a workaround that restarts the client when messages are lost. If there's no proper network solution, then this could work. Tested in my environment for 40hrs w/o any issues.

Mark323513, 2023-05-12:
I have the same issue with mz Homey 2023,

The diagnostic report has no 09307C63

SWarnberg, 2023-06-02:
@Mark323513 You can clone my PR #24 and install it on your Homey. I've run it for almost two months now without any issues.
This repo seems to be idle right now.

ricott, 2023-06-04:
Thinking out loud. Since the EM2 device is consuming UDP multicast datagrams simply by opening a port locally on the Homey - it doesn't connect to anything remote. Could these problems be based on your Homey disconnecting from your wifi from time to time and this is what causes the problem?

SWarnberg, 2023-06-04:
I wouldn't think so, since this happens exactly after 4+ minutes, every time, while I'm still connected to Homey in debug mode (or with app)..

ricott, 2023-06-05:
Historically, over the years I have developed the app, the only problems related to EM2 have been network related. Since the logic is so simple, no stateful connectivity to any external device, just open local port and accept what is pushed there.

There are 1000+ installs of the app, so I still think what you are experiencing is some local network wifi/UDP/multicast issue - otherwise a lot of users would have complained. Anyhow, I have merged your pull request.

closed 2023-06-28 by ricott