ricott/sma.modbus #26: Support for SMA Sunny Tripower 10.0 SE

TimmyT68, 2023-06-28:
Hello,

I have a Sunny Tripower 10.0 SE Inverter. And all the values are not correct reportet. The battery and the loading value  are totally not reportet.
Will you support this inverter?

Best regards
TimmyT68

ricott, 2023-06-28:
Hi!

There is a new test version, https://homey.app/a/sma.modbus/test/
I quickly enabled the recognition of the SE hybrid inverters, you should add it as an inverter device.
Won't show any battery info as of yet, but you should see all PV capabilities.

TimmyT68, 2023-06-30:
Hi,thank you. This version works fine for me. Did you support the battery in a future version also?Best regards Thomas

ricott, 2023-06-30:
Should be able to easily add more capabilities for these hybrid inverters 👍
Question is what properties are most interesting?

TimmyT68, 2023-07-01:
Hi,

I think some information about the battery conditions is interesting.

For example the actual state of charge in percent, the actual charging or decharging in W for the beginning.

Best regards
Thomas

ricott, 2023-07-02:
https://community.homey.app/t/app-pro-sma-solar-devices/696/318

closed 2023-07-21 by ricott