ricott/sma.modbus #27: can´t connect SMA Tripower Homey Pro

tkredbaron, 2023-06-29:
hey,

i can´t connect my SMA Tripower Inverter to my Homey Pro.
other App´s do (example Tibber).

(iPhone 13 Pro iOS 16.5.1)

ricott, 2023-06-29:
Can you be more precis in which inverter model you have?

tkredbaron, 2023-06-29:
Sunny Tripower 7000TL-20

ricott, 2023-06-29:
The model is supported, no problems.
Please make sure modbus is enabled and you are using the correct IP address.

tkredbaron, 2023-06-29:
dont work. see attached

ricott, 2023-06-29:
attachments don't work in email on GitHub for some reason, so don't see what you attached.
I'm still wondering though if you have made sure that modbus is enabled in the inverter?

tkredbaron, 2023-06-29:
hey,i dont know. i have found nothing at sma dashboard to manage modbus.but i think no modbus no tibber :)sma and tibber app work fine.   beste grüsse   tobias kolesnyk chefpilot marketing & design red baron marketing gmbh & co. kg+49 28 31. 977 88 12 | +49 177 - 792 55 08 | beurskensweg 1 47608 ***@***.*** | www.red-baron.de Diese E-Mail enthält rechtlich geschützte Informationen. Das unerlaubte Kopieren sowie die unbefugte Weitergabe dieser Mail ist nicht gestattet.

ricott, 2023-06-29:
Click the link I pasted above to read about how you enable modbus. You don't have to guess, just read the instructions. And no, Tibber doesn't use modbus.

closed 2023-06-29 by ricott

tkredbaron, 2023-06-29:
i cant install sunmy explorer. i have only apple mac.sunny tripower have no webinterface. i only can configure over sma sunnyportal.

tkredbaron, 2023-06-29:
> Im also a Mac user, but have an old pc for stuff like this. I am not aware of any other way to enable modbus :(

i‘m sorry. my last PC i have in 1997 😂