ricott/sma.modbus #28: SMA TRIPOWER 6.0 does not seem to be accepted.

EBlom71, 2023-07-04:
My SMA SUNNY TRIPOWER 6.0 cannot be added to Homey.

It comes with the following response:

[image]

I am sure the IP address as well as the port is correct.

ricott, 2023-07-05:
You have enabled modbus on the inverter as per instructions [here](https://community.homey.app/t/app-pro-sma-solar-devices/696#enable-modbus-in-your-inverterstorage-product-9)?

EBlom71, 2023-07-05:
Thank you Richard, now it works!Met vriendelijke groet, Erwin BlomVan: Richard Barkestam ***@***.***>Verstuurd: woensdag, juli 5, 2023 8:01:12 a.m.Aan: ricott/sma.modbus ***@***.***>CC: EBlom71 ***@***.***>; Author ***@***.***>Onderwerp: Re: [ricott/sma.modbus] SMA TRIPOWER 6.0 does not seem to be accepted. (Issue #28)
You have enabled modbus on the inverter as per instructions here?

closed 2023-07-05 by ricott