ricott/sma.modbus #29: SMA energy meter (homemanager 2.0) line voltage values

tjarb, 2023-07-11:
Dear reader,

Is it possible to add the homemanagers line voltages? They seem to be missing?

Best regards!

ricott, 2023-07-13:
Other than it gets crowded in the device card, no problems. The data is already there in the code. Will check once Im back from my vacation.

tjarb, 2023-07-13:
Hi Richard,

Thank you for your reply from from your vacation.
Our inverters switch off due to high grid voltages. The utility provider
does not act on complains, they keep pointing to the possibility of an
incorrect PV installation. So we have to prove that the high voltages occur
at the grid connection, not at the PV inverters and need the grid voltages
from the SMA energy meter.

Enjoy your vacation!

Best regards

ricott, 2023-07-21:
Available in the new [test version](https://homey.app/a/sma.modbus/test/)