ricott/sma.modbus #30: SMA Inverter Battery

MikeRow1972, 2023-07-13:
I will create a flow and use Inverter Cart "Battery charge changed to%" but the input field is missing.

Is this a fail on my side or on the card.

I use Homey pro 2023

[image]

ricott, 2023-07-13:
Currently on a trip so cant check much. But is this the sunny boy device type or do you have the new smart energy hybrid inverter?

MikeRow1972, 2023-07-13:
Hello Richard, no problem. It’s a SMA SBS 3.7-10

Regards Mike

MikeRow1972, 2023-07-13:
device details i can see 
[image]

MikeRow1972, 2023-07-22:
Hello Richard, any news about my problem? 

best regards Mike

ricott, 2023-07-22:
Came back home yesterday, so not been able to spend a lot of time yet :)

You have a sunny boy storage from what I can tell. 
I added support for the new hybrid inverters in the inverter device type recently, and I'm considering adding sunny boy storage support there as well - to phase out the storage device type.

Since I don't have a sunny boy unit it is a little problematic with testing. Can you share the values you see in addition to the ones you already shared.

MikeRow1972, 2023-07-22:
Hi Richard, I'm a Homey newbie but I work in the IT industry. I think you can guide me through the system for troubleshooting.
In the device properties I see the following values + Battery SOC

[image]

and following cards 
[image]
[image]

This when Cards has no input fields

[image]

Is that what you need from me?

ricott, 2023-07-22:
Thanks for the screen shots

As for the support for hybrid/battery inverters in the normal inverter device type;
- I have combined charge and discharge into one capability for the hybrid inverters, value is negative for discharge.
- Battery state of charge is using the Homey built in battery capability, shown on a separate tab.
- I have just added the battery operational status, not released yet.

The capabilities I'm interested in are these. 
- power drawn
Not sure what this actually shows? Different from discharge?
- power grid feedin
My thinking is that any grid values require an energy meter or home manager and these values are available in that device
- battery capacity
What is battery capacity vs state of charge (SOC)? Both value are in % as per the spec.

MikeRow1972, 2023-07-22:
power drawn
Not sure what this actually shows? Different from discharge?

the value is correct, battery is full 
[image]

power grid feedin
My thinking is that any grid values require an energy meter or home manager and these values are available in that device

the value is correct 
[image]

battery capacity
What is battery capacity vs state of charge (SOC)? Both value are in % as per the spec.
values are correct 
capacity 96%
state of charge 100% 

[image]

Do you know why the input fields from Flow cards missing??

ricott, 2023-07-22:
You can now try the new [test version](https://homey.app/a/sma.modbus/test/)
Add the Sunny Boy Storage as an `inverter` device type, not `storage` device type.

MikeRow1972, 2023-07-23:
Hello Richard, sounds good. 
[image]
[image]
[image]

closed 2023-12-13 by ricott