ricott/sma.modbus #31: Request: limit inverter output

tkroon89, 2023-08-15:
Hi Richard!

It’s not an issue, but a question. Would it be possible to limit the inverter output via modbus/your app? I read it on this page:

https://www.photovoltaikforum.com/thread/173823-limit-sunny-tripower-via-modbus-based-on-grid-injection/

Of course I can help you test or investigate if you’re interested :)

ricott, 2023-08-18:
Hi!

This is for sure an interesting topic given the fluctuations in electricity prices as of late.
I recently added an action for limiting the inverter output, `Limit the active power out to [power]`. I have tried this myself and it works fine. This uses modbus registry `40915`.

It was mentioned in the App forum thread that this setting writes to flash storage inside the inverter and should not be changed excessively (~100k writes 🤷‍♂️). I have not verified this though.

I myself have a Home Manager 2 and would love to find a way to be able to set using a Homey action zero export when electricity prices are negative. Not found a way to do that though.

Any intel into how we could achieve this is very welcomed 😀

tkroon89, 2023-08-18:
Awesome, I wasn’t aware this already existed. I tried it and it worked nicely! If I limit it to 0, the result is 10W. Close enough I think. 

I also read the limit of writes, so better to not change it every minute. But the moments the energy tariffs get negative this will do the trick. Thanks!!

closed 2023-10-09 by ricott