ricott/sma.modbus #33: Request: configurable IP on the EnergyMeter (Home Manager 2)

elpaha, 2023-11-11:
I've been a happy user of this homey app for quite some time, kind thanks for your efforts.  The phase-reporting on the energy meter is a great help when analysing/optimising/balancing load. 

However discovering/connecting to the Home Manager can be quite a hassle, if not impossible, in extended networks or routing-situations that do not support multicast. For inverters there is a configurable IP address, but not for the Home Manager 2. 
Would it be possible to add this parameter to the EnergyMeter too?

Regards

ricott, 2023-11-13:
Hi!

Currently the Home Manager implementation is based on multicast (as you also state). I have not seen any other implementation for it using modbus for instance.

I however stumbled upon a config option in Sunny Portal which potentially could help (haven't tested myself).
1. Log into Sunny Portal
2. Device Overview
3. Home Manager Properties
4. Edit button
5. Choose Complex Configuration at top
6. Under Meter Configuration, specify your Homey's IP address. As the info says it will use unicast to that IP.

ricott, 2026-10-24:
Closing: the Home Manager only provides data via multicast (Speedwire), so there is no IP to configure. As a workaround, the Home Manager can be configured in Sunny Portal to send unicast data directly to Homey's IP address.

closed 2026-10-24 by ricott