ricott/sma.modbus #35: New inverter code

BusterXXL, 2023-12-13:
Not really an issue, but just discovered Your great app for Homey and installed it and got it to search for my SMA Tripower X 15 which it found. Tho, it recognize it  as "Unknown" since it doesn't recognize the type id for it.
The model is a SMA Tripower X 15 / STP 15-50 and it reports type id "9491".
So it would be nice if could be added to the switch block in decodeDevice function :)  (worked as  C/C++ developer for 30+ years, tho I'm not a js-coder but I can  probably manage to do the change my self and push it and create a pull request for it if You prefer that)
I also use the app to get data from a SMA Energy Meter (EMETER20) that is "handled" by the Tripower X 15, and the app works like a charm for that also !!

FYI: Modbus spec for the SMA Tripower X XX/STP XX-50 can be found here.

https://files.sma.de/downloads/STP-X_STP_xx-50_Sunspec_Modbus_Model_PICS_Overview_V1.zip

If You want me to verify anything on the SMA XX-50 model, just give me shout :)

BusterXXL, 2023-12-13:
Worked like a charm, with out having to remove/add the inverter 👍 
Tho, software version it displays seems incorrect, should be 03.02.08.R but the app shows 165.65.55.255
See attached screenshot.
[image]

ricott, 2023-12-13:
Good that it worked!
Strange with the software version. As per the spec it should be Modbus register `30059`, the same as all other inverters. The value here looks like an ip address ...

[image]

BusterXXL, 2023-12-13:
Just saw that the voltages on the AC side does not work in the test version.
only L1 and L2 is listed and both got 0 values with the test version (had values before according to insights, tho can't see any of those parameters in the "instantaneous values" either in the tripowers web GUI for the moment. No production due to snow on the solar panels).

Not 100% sure that the link to the modbus parameters for the tripower is the most recent one.
Attaching an archive which has list of modbus parameters for quite recent Tripower firmware version
[PARAMETER-STPxx-50_03-02-07-R.zip](https://github.com/ricott/sma.modbus/files/13661686/PARAMETER-STPxx-50_03-02-07-R.zip)

ricott, 2023-12-13:
There is a [BASE](https://github.com/ricott/sma.modbus/blob/master/lib/inverters/base.js) config in the bottom for all inverters.
Before this test version, the Tripower should have worked as a [BASIC](https://github.com/ricott/sma.modbus/blob/master/lib/inverters/basic.js) type - the least number of capabilities.
Now it should be interpreted as a [TRIPOWER](https://github.com/ricott/sma.modbus/blob/master/lib/inverters/tripower.js) version.

These other types are added on top of the BASE config, and can change or simply remove capabilities. BASIC has ac voltage for L1 only, TRIPOWER should have for all three phases.

If you for instance don't see a daily yield capability, then you need to remove the device and re-add it.

dolf1966, 2023-12-19:
[image]

[image]

It's not working, can you help

ricott, 2023-12-19:
@dolf1966 Your problem is not related, so move it to a new ticket ....

closed 2023-12-20 by ricott