ricott/sma.modbus #37: SI 6.0 inverter not working

ricott, 2023-12-19:
[image]

[image]

It's not working, can you help

_Originally posted by @dolf1966 in https://github.com/ricott/sma.modbus/issues/35#issuecomment-1863221265_

ricott, 2023-12-19:
As per my old docs since maaany years SMA's Modbus spec for this inverter type, `SI n.n...` doesn't contain Modbus IDs - thus it is tough to program anything around it.

ricott, 2023-12-20:
I did some changes related to sunny island inverters, please try the latest[ test version](https://homey.app/a/sma.modbus/test/).

dolf1966, 2023-12-20:
hello Richard. I can add it now. only a few modbus addresses are incorrect.
When I get home I take a few screenshots of the modbus addresses I use.
 hopefully I can help with that .

Spreek je ook Nederlands?
Dolf.

dolf1966, 2023-12-20:
maybe this help
[image: image.png]

ricott, 2023-12-20:
Sorry, images doesnt work in email to github. Can you upload it here directly?

dolf1966, 2023-12-20:
[image]

[image]

ricott, 2023-12-21:
This is the battery modbus registry id's currently for `SI4.4M-12 / SI6.0H-12 / SI8.0H-12 / SI4.4M-13 / SI6.0H-13 / SI8.0H-13 (SUNNY ISLAND 4.4M / 6.0H / 8.0H)` inverters. 
(I know from the screenshot you have a SI6.0H-1**1** - eleven, not twelve or thirteen. So maybe there is a difference...)

```
batterySoC: new ModbusRegistry(setting.READING, 30845, type.U32_FIX0, 'Current battery state of charge', 'measure_battery'),
batteryCharge: new ModbusRegistry(setting.READING, 31393, type.U32_FIX0, 'Present battery charge', 'measure_power.battery'),
batteryDischarge: new ModbusRegistry(setting.READING, 31395, type.U32_FIX0, 'Present battery discharge', 'measure_power.battery'),
batteryStatus:  new ModbusRegistry(setting.READING, 30955, type.BATTERY_STATUS, 'Battery operating status', 'operational_status.battery')
```

This from the SMA modbus spec
[image]
[image]
[image]

30845 we have in common, and value of that is shown on the battery tab on the device. 
[image]

Will need to look deeper into the other three values.
Do you have an sma energy meter connected directly to your inverter?

dolf1966, 2023-12-21:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/HoDo/AppData/Local/Packages/oice_16_974fa576_32c1d314_ad/AC/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/HoDo/AppData/Local/Packages/oice_16_974fa576_32c1d314_ad/AC/Temp/msohtmlclip1/01/clip_filelist.xml">
<title>Device-specific Modbus adresses</title>
<style>

</style>
</head>

<body link=blue vlink=purple>

SMA Modbus Profile |   |  
-- | -- | --
Modbus register address | Short description | Type | Format | Unit | Access | Scale factor |   | Supported hardware and firmware:
30051 | Device class | U32 | ENUM |   | RO |   |   | Sunny Island with "Speedwire data module Sunny   Island"
30053 | Device type | U32 | ENUM |   | RO |   |   | Sunny Island: SI 6.0H-11 and SI 8.0H-11
30057 | Serial number | U32 | RAW |   | RO |   |   | Speedwire data module Sunny   Island: SWDMSI-NR10
30059 | Software package | U32 | FW |   | RO |   |   | Starting with software package: 3.100
30197 | Current event number | U32 | FIX0 |   | RO |   |   |  
30199 | Waiting time until feed-in | U32 | Duration | s | RO | 1 |   |  
30201 | Condition | U32 | ENUM |   | RO |   |   |  
30211 | Recommended action | U32 | ENUM |   | RO |   |   |  
30213 | Message | U32 | ENUM |   | RO |   |   |  
30215 | Fault correction measure | U32 | ENUM |   | RO |   |   |  
30559 | Number of events for user | U32 | FIX0 |   | RO |   |   |  
30561 | Number of events for installer | U32 | FIX0 |   | RO |   |   |  
30563 | Number of events for service | U32 | FIX0 |   | RO |   |   |  
30565 | Number of generator starts | U32 | FIX0 |   | RO |   |   |  
30567 | Amp hours counter for battery charge | U32 | FIX0 | Ah | RO |   |   |  
30569 | Amp hours counter for battery   discharge | U32 | FIX0 | Ah | RO |   |   |  
30571 | Meter reading consumption meter | U32 | FIX0 | Wh | RO |   |   |  
30577 | Energy consumed from grid (today) | U32 | FIX0 | Wh | RO |   |   |  
30579 | Energy fed into grid (today) | U32 | FIX0 | Wh | RO |   |   |  
30581 | Grid reference counter reading | U32 | FIX0 | Wh | RO |   |   |  
30583 | Grid feed-in counter reading | U32 | FIX0 | Wh | RO |   |   |  
30585 | Power outage | U32 | Duration | s | RO | 1 |   |  
30587 | PV generation counter reading | U32 | FIX0 | Wh | RO |   |   |  
30589 | Rise in self-consumption | U32 | FIX0 | Wh | RO |   |   |  
30591 | Rise in self-consumption today | U32 | FIX0 | Wh | RO |   |   |  
30593 | Energy consumed internally | U32 | FIX0 | Wh | RO |   |   |  
30595 | Absorbed energy | U32 | FIX0 | Wh | RO |   |   |  
30597 | Released energy | U32 | FIX0 | Wh | RO |   |   |  
30599 | Number of grid connections | U32 | FIX0 |   | RO |   |   |  
30775 | Power | S32 | FIX0 | W | RO |   |   |  
30777 | Power L1 | S32 | FIX0 | W | RO |   |   |  
30779 | Power L2 | S32 | FIX0 | W | RO |   |   |  
30781 | Power L3 | S32 | FIX0 | W | RO |   |   |  
30783 | Grid voltage phase L1 | U32 | FIX2 | V | RO |   |   |  
30785 | Grid voltage phase L2 | U32 | FIX2 | V | RO |   |   |  
30787 | Grid voltage phase L3 | U32 | FIX2 | V | RO |   |   |  
30797 | Grid current phase L1 | U32 | FIX3 | A | RO |   |   |  
30799 | Grid current phase L2 | U32 | FIX3 | A | RO |   |   |  
30801 | Grid current phase L3 | U32 | FIX3 | A | RO |   |   |  
30803 | Grid frequency | U32 | FIX2 | Hz | RO |   |   |  
30805 | Reactive power | S32 | FIX0 | VAr | RO |   |   |  
30807 | Reactive power L1 | S32 | FIX0 | VAr | RO |   |   |  
30809 | Reactive power L2 | S32 | FIX0 | VAr | RO |   |   |  
30811 | Reactive power L3 | S32 | FIX0 | VAr | RO |   |   |  
30825 | Operating mode of stat.V stab.,   stat.V stab. config. | U32 | ENUM |   | RO |   |   |  
30831 | cosPhi setpoint, cosPhi config.,   direct specif. | S32 | FIX2 |   | RO |   |   |  
30833 | cosPhi excit.type, cosPhi config.,   direct spec. | U32 | ENUM |   | RO |   |   |  
[... 247 lines cut ...]
40769 | Lmt value battery state of charge for   start load shedding 2 | U32 | FIX0 | % | RW |   |   |  
40771 | Lmt value battery state of charge for   stop load shedding 2 | U32 | FIX0 | % | RW |   |   |  
40773 | Start time additional time range load   shedding 2 | U32 | TM |   | RW |   |   |  
40775 | Time load shedding 2 | U32 | TM |   | RW |   |   |  
40777 | Limit battery state of charge for   start load shedding 2 in add time range | U32 | FIX0 | % | RW |   |   |  
40779 | Limit battery state of charge for   stop load shedding 2 in add time range | U32 | FIX0 | % | RW |   |   |  
40781 | Temp lmt for multifunct. relay with   batt comp fan | S32 | TEMP | °C | RW | 0.1 |   |  
40783 | Serial no. Slave1 (Phase L2) | U32 | RAW |   | RO |   |   |  
40785 | Serial no. Slave2 (Phase L3) | U32 | RAW |   | RO |   |   |  
40787 | Cluster behaviour when a device fails | U32 | ENUM |   | RW |   |   |  
40789 | Communication version | U32 | REV |   | RO |   |   |  
40791 | Timeout for communication fault   indication | U32 | FIX0 | s | RW | 1 |   |  
40795 | Max. battery charge capac. | U32 | FIX0 | W | WO |   |   |  
40799 | Max. battery discharge capac. | U32 | FIX0 | W | WO |   |   |  
40805 | Energy saving mode | U32 | ENUM |   | WO |   |   |  
40811 | Update version of the logic component | U32 | FW |   | RO |   |   |  
40819 | Firmware version of protocol   converter | U32 | FW |   | RO |   |   |  
40821 | Hardware version of protocol   converter | U32 | HW |   | RO |   |   |  
40827 | Serial number of the protocol   converter | U32 | RAW |   | RO |   |   |  
40829 | SUSyID of protocol converter | U32 | FIX0 |   | RO |   |   |  
43090 | Login with Grid   Guard-Code | U32 | FIX0 |   | RW |   |   |  

</body>

</html>

dolf1966, 2023-12-21:
i have the sma electric meter with utp conection (ip-adres, no modbus)

tjarb, 2024-10-16:
would also love to have a status of the grid connection or inverter operation state.
Like grid mode and offgrid mode (or island or backup mode).

ricott, 2026-10-24:
The SI 6.0 issue was fixed back in Dec 2023. The grid/off-grid status request is tracked in #45.

closed 2026-10-24 by ricott