ricott/sma.modbus #38: SMA Home Manager cannot be connected anymore.

lasertack, 2024-01-18:
Hi!

Few days ago the SMA Manager gone offline within the homey app. SMA App still works fine. 
Homey stopped getting data from it. 
I deleted it and tried to add again but now I dont see it anymore and get "Cannot read properties of null (reading length)"

Any idea how to solve it?

Thanks

Alex

ricott, 2024-01-22:
Hi!

The Home Manager integration is based on UDP multicast. That is the home manager sends out datagrams to the entire network using multicast and the Homey app just listens to them on a predefined port. If the integration stops working, then the SMA Home Manager's UDP multicast datagrams no longer reach the network where the Homey is located.

Both the add new device process and the normal operations process are the same, open a local port on the Homey and listen for UDP datagrams.

Anything that changed in your network recently?

lasertack, 2024-01-22:
Hi! Thank you so much for your reply. Yes, there was some change. The whole IoT has been moved to a separate network but Homey didn’t have access to this network. That’s why it stopped communicating. Today I was able to solve it. Thanks a lot!Alex

closed 2024-01-22 by ricott