ricott/sma.modbus #40: Future Request: Sporadic "Failed to Read" for inverter (eg. Tripower STP 6000TL-20) - please bypass the timeout

josefzahner, 2024-02-16:
I own two different SMA Tripower Inverters (STP 6000TL-20 & STP 7000TL-20). Both have the issue that the communication on the inverters itself is blocked for about 15s every 5min or so. They completely refuse to communicate during that time. The problem is, during the communication timeout, the SMA homey app sets all values of the inverter to "0" or default as soon as (I'm guessing) one poll interval was done without a response, which is quite irritating as I'm reading out the values and present them on a Ulanzi TC001 matrix display - so it happens a lot that I'm checking the display and it says zero solar production (which in fact is not true). 
**Now to my idea, would it be possible to keep the last read values in case of the "Failed to Read" ( short communication drop on SMA inverter) for lets say 30s or 60s instead of reseting them to zero? Or even better, create a dedicated option in the app where you can specify the retry count until it zeroing out everything**
I know I could create Homey variables to workaround the issue, but I own two inverters and the logic would get a bit complex to ensure that I'm doing it right just to skip the wrong 0 values.

Below some details to the connection error.

Homey SMA App Error: 
[image]

Wireshark Capturing:
I do see that homey (192.168.20.203) tries to read modbus register 30771 (192.168.20.182) on one of the inverters, but homey doesn’t get any answer for more 10 seconds
[image]

jozahner, 2025-05-14:
A few months ago I switched to Homey Pro 2023 (via LAN Cable connected) and the errors are more or less gone...

ricott, 2026-10-24:
Addressed in v2.9.3–v2.10.1: configurable Modbus timeout, automatic connection recovery, failed reads no longer reset values to zero, and devices wait 5 minutes before being marked unavailable.

closed 2026-10-24 by ricott