ricott/sma.modbus #42: Sunny Boy 1300TL-10 - not able to find inverter

Otterkop, 2024-06-17:
Hi Ricott,

I have my Sunny Boy 1300TL-10 installed since May 2014. Just now I am considering connecting it to Homey Pro. However, I am not that lucky. 

I entered my IP-address to the app during configuration and I get the message 'Wrong IP number or port, no SMA inverter found'. I am not sure what to do. I tried to read through some other threads but they are somewhat too technical for me. 

What can I try / do to solve my issue?

Thanks! Hans

josefzahner, 2024-07-06:
Ensure that modbus TCP is enabled. Per default it isn‘t…

Otterkop, 2024-07-10:
Thanks Josef. And I guess this means that my TL1300 is not supported. Which clarifies a lot because I couldn't find any documentation on how to enable TCP on my Sunny Boy.

josefzahner, 2024-07-10:
Did you check the sunny explorer with the installer account for your inverter for modbus tcp like in the explanation of the forum?

Otterkop, 2024-07-11:
Mostly I am very content with having a MacBook. In very few occasions I get frustrated because I must use a PC-executable ;-). I did not try the explorer yet. You seem to suggest that this explorer could do the job. Possibly I will then install Parallels to just find out. Thanks!!

Otterkop, 2024-07-12:
Installed Parallels and the Sunny Explorer. Followed instructions - no joy. Sunny explorer doesn't find any plant :-(

closed 2025-06-07 by ricott