ricott/sma.modbus #45: Feature request - External power connection voltage

josjg, 2024-10-15:
Hello Richard,

Just started using your app on my SMA installation. I am working in an off-grid setup with batteries, 3 sunny Island 8.0H-13 inverters and a Tripower 8.0 solar inverter. This is because the utility grid in my area is unreliable and has large voltage fluctuations. 

The system is set up to connect to the utility gris when the batteries fall below 20% charge, and then they get charged by the utility grid through the sunny islands to 30%. This  in case of long periods without sun. The utility gris is used as a generator, so to speak.

When the utility grid is not available, which happens sometimes, the batteries cannot be charged in this situation and I would like to automatically cut large consumers. For that I would need a card that tells me that the utility gris connection has low voltage.

My installer tells me that this data is available in the sunny island, under "external Power Connection -> Phase Voltage".
Would this data also be available to your app? And if so could you add a card with the logic to activate when the voltage falls below a certain value?

That would be great!

Best regards, Jos

josjg, 2024-10-15:
Hello again,

I did some investigation en experimenting myself and found that these external voltages values are in registries 30903, 30905 and 30907 respectively for L1,L2,L3. 
The amperes are in the next 3 (odd numbered) registers and status of the external connection is 30915.
I have verified that these voltages drop when I disconnect the utility grid manually.

Hope that helps, and hope that you will include these data, and cards for them!

Thanks in advance, Jos

[SMA_Modbus-SI inverter.pdf](https://github.com/user-attachments/files/17382008/SMA_Modbus-SI.inverter.pdf)

ricott, 2026-10-24:
Added in v2.11.0. For Sunny Island devices added as a **Battery** device:
- External grid voltage per phase (registers 30903/30905/30907) and grid connection status (30915: Mains connected / Backup / Backup not available / Off).
- Flow cards: trigger *Grid voltage on [phase] dropped below X V* (fires once when crossing the threshold), trigger *Grid connection status changed*, and conditions *Grid voltage is below X V* and *Grid connection status is …*.

Note: on single-phase off-grid systems the unused phases read 0 V, so pick the specific phase rather than 'any phase' in the voltage cards. Please reopen if something doesn't work on your Sunny Islands.

closed 2026-10-24 by ricott