ricott/sma.modbus #46: SMA "SE inverters" + "home storage batteries" are still missing to be connected to Homey.

webmilesking, 2024-12-20:
The most important SMA "SE inverters" and "home storage batteries" are still missing to be connected to Homey. 
Only data from home manager 2.0 works. 

**It's not yet possible to get any data from SE inverters:  Hybrid STP 10.0-3SE-40 and connected home storage batteries**

ricott, 2025-01-21:
The SE inverters are supported and should be recognised with a battery for instance.
How are you adding the inverter and what do you see today?

SvenWS, 2025-03-03:
All storage inverters like Sunny Boy should be classified as battery and not as sensor or solarpanel. In the energy section these devices should have set the  homeBattery flag set to true.

The driver.compose.json of storage devices should look like

```
...
  class: 'battery',
...
  energy: { batteries: [ 'INTERNAL' ], homeBattery: true },
...
```
I made a poc by overwriting these properties and got the storage correctly displayed in the new energy tab.

jeroenhub1, 2025-08-12:
Until version 2.7.6 my STP SE works fine, show all relevant information including battery level

ricott, 2025-08-12:
https://community.homey.app/t/app-pro-sma-energy-my-home-my-energy-our-climate/696/454?u=richard_b

ricott, 2026-10-24:
Closing: SE inverters are supported, and home storage batteries now have a dedicated Battery device type that works with Homey Energy.

closed 2026-10-24 by ricott