ricott/sma.modbus #47: Integration of Sunny Tripower 6.0 SE

fhlznr, 2025-01-21:
Can you integrate the  Sunny Tripower 6.0 SE device so that I can also receive additional data from the PV system in Homey, such as battery storage, storage charge, current power generation via the PV system, etc.?

ricott, 2025-01-21:
The device is supported and should be recognised with a battery for instance.
How are you adding the inverter and what do you see today?

fhlznr, 2025-01-22:
Hi Richard,i only see in homey my sunny Home Manager 2.0 and the following Information( attachement)Home can i setup the sunny tripower 6.0 se?Gesendet mit der mobilen Mail AppAm 21.01.25 um 20:46 schrieb Richard Barkestam
                
            
            
                
                    Von: "Richard Barkestam" ***@***.***>Datum: 21. Januar 2025An: "ricott/sma.modbus" ***@***.***>Cc: "fhlznr" ***@***.***>,"Author" ***@***.***>Betreff: Re: [ricott/sma.modbus] Integration of Sunny Tripower 6.0 SE (Issue #47)
                
                
The device is supported and should be recognised with a battery for instance.
How are you adding the inverter and what do you see today?

ricott, 2025-01-22:
You add an `inverter` device in the same way you added an `energy meter` device to get the home manager into Homey.

fhlznr, 2025-01-22:
I only Need the iP address of the device and choose as device „inverter“?Gesendet mit der mobilen Mail AppAm 22.01.25 um 13:03 schrieb Richard Barkestam
                
            
            
                
                    Von: "Richard Barkestam" ***@***.***>Datum: 22. Januar 2025An: "ricott/sma.modbus" ***@***.***>Cc: "fhlznr" ***@***.***>,"Author" ***@***.***>Betreff: Re: [ricott/sma.modbus] Integration of Sunny Tripower 6.0 SE (Issue #47)
                
                
You add an inverter device in the same way you added an energy meter device to get the home manager into Homey.

ricott, 2025-01-22:
If you allow multicast in your network you don't need anything, it will be auto-detected.
Please note that all SMA inverters have Modbus turned off by default; you need to [enable it first](https://community.homey.app/t/app-pro-sma-solar-devices/696).

ricott, 2026-10-24:
Closing as answered: STP SE is supported. Enable Modbus on the inverter, then add it as an Inverter device and the battery as a Battery device.

closed 2026-10-24 by ricott