ricott/sma.modbus #48: Battery values of measure_power

SvenWS, 2025-03-03:
If storage inverters correctly classified as batteries, the value of measure_power (labeled with Grid Power) should be inverted: loading (charging) should have a positive value, delivering energy should be negative. 

This is doumented here:
https://apps.developer.homey.app/the-basics/devices/energy#home-batteries

ricott, 2025-03-03:
I realize you wrote [this comment](https://community.homey.app/t/app-pro-sma-solar-devices/696/417) in the forum thread.
You have an older generation battery inverter, sunny boy storage, which you have added using the inverter device type?

That would explain https://github.com/ricott/sma.modbus/issues/50. It ends up not summarizing solar power, but mixing solar power and battery power.

To complicate things further, the new hybrid inverters, Sunny Boy Smart Energy, combine solar panel and battery device classes in a single device—something that Homey Energy does not support. I am starting to think I need to separate this into two different device types to make it work. I just now did this in the Victron app.

SvenWS, 2025-03-04:
I will try to re add the device with the right device type storage. Last time it just popped up as new device. And yes at that time it was classified as solarpanel and not as storage.

(Right now I just have remote access and forgot the ip address. I will onsite latest next week and give it a try.)

SvenWS, 2025-03-04:
If I add the Sunny Boy Storage 5 as Sunny Boy Storage nothing happens. No values are filled in.

If I add the SBS 5 as inverter values are filled in.
After that I changed the class and energy.

SvenWS, 2025-03-05:
To keep it simple: why not detect SBS in the device type string. If so treat the device as home battery and just overwrite measure_power with measure_power.battery. Then classification and grid power would be right for homey energy.

SvenWS, 2025-03-07:
I looked to code and saw that this detection is already done :-). But the three device types are treated equally. So I would be easy to to change the class and set home battery flag on device type 8007?!

ricott, 2026-10-24:
Resolved: storage/batteries now have a dedicated Battery device type (class battery, homeBattery) that works with Homey Energy. Please add Sunny Boy Storage via the Battery device type.

closed 2026-10-24 by ricott