ricott/sma.modbus #49: measure_power of sensor

SvenWS, 2025-03-03:
The measure_power of a sensor should have negative values if feeding the grid. Right now it has a zero value and the measure_power.surplus has a positive value. This will lead to a more acurate daily graph in the energy tab.

ricott, 2025-03-03:
Today, it probably makes sense; it will be a breaking change - but maybe worth it.

SvenWS, 2025-03-04:
Why not use a feature toggle un the UI to avoid breaking changes?

ricott, 2025-03-04:
It shouldn't be a biggie. Just published a [test version](https://homey.app/a/sma.modbus/test/) with the change.

SvenWS, 2025-03-04:
Because I deleted my SBS and can't remember the IP I have to be onsite to test. That will be at the weekend.

SvenWS, 2025-03-04:
I can add the SBS 5 as SMA Inverter and the properties are filled. But it has then the class solarpanel. 
If I register the SBS 5 as Sunny Boy Storage the properties are not filled.

closed 2025-06-07 by ricott