ricott/sma.modbus #50: measure_power.consumption and measure_power.pv of summary

SvenWS, 2025-03-03:
In the case that a battery plays a role the values of  measure_power.consumption and measure_power.pv are not correct.
I attached two pics.

[image]

[image]

ricott, 2025-03-03:
Consumption could obviously be wrong since that is calculated. 
I'm unsure how PV can be wrong, though, since it only summarizes `measure_power` from the `inverter` device type.
`power_pv = power_pv + inverter.getCapabilityValue('measure_power');`
Does the measure_power of your inverter device display correctly?

SvenWS, 2025-03-04:
I will te add the device and asure that it will be registered as storage.

ricott, 2026-10-24:
Resolved: batteries are now a separate Battery device type and the Energy Summary takes battery power into account when calculating PV and consumption.

closed 2026-10-24 by ricott