ricott/sma.modbus #51: Second SMA inverter add to Homey

frank29081974, 2025-03-07:
I have two SMA inverters 
I managed to add one of the inverters in Homey in the SMA energy app (modbus in both inverters is set to yes). Homey found it immediately without asking for an IP address. 
Now I would like to add the other inverter in Homey. Homey asks for 192.168. 178. ... but then I get the message wrong IP address. 
The IP address of the inverter is 192.168. 0.....
In the Ziggo modem I cannot add the inverter as a fixed IP address 192.168. 178.... 
Does anyone know how to add the 2nd SMA inverter to Homey?

mathijsm0, 2025-03-09:
I experience the same issue, after installing the first, I was not able to add a second configuration (yes, modbus on).

ricott, 2025-03-10:
@frank29081974 
> Homey asks for 192.168. 178. ... but then I get the message wrong IP address. The IP address of the inverter is 192.168. 0.....
This I don't follow, "Homey asks for"? As the add new inverter page says, either it is found automatically - i.e. you have Modbus turned on and your local network has multicast enabled - or you manually enter the IP address of the inverter. The app never gives a proposal for an IP address.

@mathijsm0 
Please send the error message and trigger a diagnostic report immediately after attempting to add the inverter.

frank29081974, 2025-03-10:
> [@frank29081974](https://github.com/frank29081974)
>
>
> This I don't follow, "Homey asks for"? As the add new inverter page says, either it is found automatically - i.e. you have Modbus turned on and [...]

In myhomey (windows) I could add the second inverter in devices, without asking for an IP address of the inverter. It works fine now. Thank you for your response.

mathijsm0, 2025-03-19:
> [@frank29081974](https://github.com/frank29081974)
>
>
> This I don't follow, "Homey asks for"? As the add new inverter page says, either it is found automatically - i.e. you have Modbus turned on and [...]

Hi Ricott,

Guess what? Just as Frank, now I have no problem adding the second inverter. Don't know how or what, but it just works. 

Thanks for your inquiry.

Petertje69, 2025-05-08:
Hello Ricott,

I have the same problem, my first inverter, a stp se,was no problem to install.
My second inverter, a stp tl30, is not found by the app.
Not on my ipad and not on a windows desktop.
When I put the IPadress and press "verbind" nothing happends.
If I try to connect it if like it is a sunnyboy storrage, it's connecting, but then it's giving no power details.

Can yoy help me connecting my second inverter?

Peter

ricott, 2026-10-24:
Closing: both original reporters got the second inverter added. If auto-discovery fails, add it with the IP address (v2.10.4 improved manual pairing and error messages). Please open a new issue with a diagnostic report if needed.

closed 2026-10-24 by ricott