ricott/sma.modbus #52: App crashes, unable to add inverter

M230, 2025-05-03:
Since updating to 2.7.0 I have experienced the app crashing at seemingly random moments. I've installed via CLI and am unable to add my inverter. Manual entry of the IP and clicking connect results in the following error:

```
2025-05-03T09:49:35.695Z [log] [ManagerDrivers] [Driver:inverter] Inverter client connected on IP '192.168.1.17'
2025-05-03T09:49:38.708Z [log] [ManagerDrivers] [Driver:inverter] Client closed for IP '192.168.1.17'
UserRequestError {
  err: 'Offline',
  message: 'connection to modbus server closed',
  request: ModbusTCPRequest {
    _id: 4,
    _protocol: 0,
    _length: 6,
    _unitId: 3,
    _body: ReadHoldingRegistersRequestBody {
      _fc: 3,
      _start: 30059,
      _count: 2
    }
  },
  response: undefined
}
2025-05-03T09:49:38.718Z [log] [ManagerDrivers] [Driver:inverter] Failed to read inverter properties Error: Failed to read '30059' for device type 'STP 9000TL-20'
    at modbusReading (/lib/sma.js:231:24)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async Promise.all (index 2)
UserRequestError {
  err: 'OutOfSync',
  message: 'rejecting because of earlier OutOfSync error',
  request: ModbusTCPRequest {
    _id: 5,
    _protocol: 0,
    _length: 6,
    _unitId: 3,
    _body: ReadHoldingRegistersRequestBody {
      _fc: 3,
      _start: 30231,
      _count: 2
    }
  },
  response: undefined
}
2025-05-03T09:49:38.728Z [log] [ManagerDrivers] [Driver:inverter] Failed to read inverter properties Error: Failed to read '30231' for device type 'STP 9000TL-20'
    at modbusReading (/lib/sma.js:231:24)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async Promise.all (index 3)
UserRequestError {
  err: 'OutOfSync',
  message: 'rejecting because of earlier OutOfSync error',
  request: ModbusTCPRequest {
    _id: 6,
    _protocol: 0,
    _length: 6,
    _unitId: 3,
    _body: ReadHoldingRegistersRequestBody {
      _fc: 3,
      _start: 40109,
      _count: 2
    }
  },
  response: undefined
}
2025-05-03T09:49:38.735Z [log] [ManagerDrivers] [Driver:inverter] Failed to read inverter properties Error: Failed to read '40109' for device type 'STP 9000TL-20'
    at modbusReading (/lib/sma.js:231:24)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async Promise.all (index 4)
2025-05-03T09:49:38.742Z [log] [ManagerDrivers] [Driver:inverter] Adding to devices: STP 9000TL-20
```

I have also tried 2.6.7, which works fine.

ricott, 2025-06-07:
By coincidence, I had to restart my inverter today to get Modbus functioning again 🤷‍♂️
I'll release a new test version with a revised implementation of the reconnect logic, amongst other things.

M230, 2025-06-09:
Restarting my inverter did not work for me. It seemed that the modbus connection was getting saturated. I rewrote most of the driver and now it works. Not sure why this started happening, it always worked fine.