ricott/sma.modbus #54: SMA SE power limit

jeroenhub1, 2025-05-26:
When trying to set a power limit on a STP SE the app gives an error: "A Modbus Exception Occured - See response body"

Xibel, 2025-06-02:
I have the same error and did some investigation. I am able to set the slider exactly down to 50% which is a value of 2500W. My SMA is a 5000W model.

If I login to my Sunny Tripower 5.0 SE as installer and look at some device parameters I see the following things.
Device > Inverter > Set active power limit is set to 5000w. The strange thing is that I cannot set this back to 0W. This is the value that the Homey App uses to set a lower value when using the slider. This parameter has an instruction of (2,500 W … 5,050 W). This tell's me there is a limit at 2500 watts. I think if I would be able to lower this value to 0 the slider is also able to set the output to 0.

I have no clue how to lower this 'Set active power limit' or some other parameter to accept values lower than 50% / 2500 watts.

At the parameter 'Operating mode active power setting' is says 'External active power setpoint'. I can change this to manual setting in % or watts. Makes no difference for Homey. I am still able to lower output to just 50% (2500watts).
 
You can also set some 'Operating mode of power control module' to 'Active power limitation' but all do not impact the slider in Homey SMA App.

If someone has an idea we would probably solve this issue with the power limit slider in the Homey SMA App.

ricott, 2026-10-24:
The 50% floor comes from the inverter itself ('Set active power limit' range 2,500–5,050 W), not the app. Since v2.9.0 there is a flow action 'Set active power curtailment (%)'. It writes register 40023, which is meant for repeated/automated control and does not wear the flash memory. External control must be enabled and a fallback configured in the inverter (installer access). Please open a new issue if that doesn't help.

closed 2026-10-24 by ricott