ricott/sma.modbus #56: devices freeze/get stuck

tjarb, 2025-07-04:
devices hang with modbus errors. only a app restart solves this.

ricott, 2025-07-04:
Need more details. Which device type?
Have you tried to send a diagnostic report?

ricott, 2025-07-05:
We have one app bug, here I blame entirely Cursor which messed this up in a refactoring 🙃
```
2025-07-05T03:33:11.412Z [err] [ManagerDrivers] [Driver:inverter] Error: Could not trigger Flow card with id "inverter_status_changed": Invalid value for token inverter_status. Expected string but got undefined
    at FlowCardTriggerDevice.trigger (/node_modules/@athombv/homey-apps-sdk-v3/lib/FlowCardTriggerDevice.js:58:17)
    at InverterDriver.triggerInverterStatusChanged (/app/drivers/inverter/driver.js:22:45)
    at InverterDevice._handlePropertyTriggers (/app/drivers/inverter/device.js:316:31)
    at InverterDevice._updateProperty (/app/drivers/baseDevice.js:53:28)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async InverterDevice.updateOperationalStatus (/app/drivers/inverter/device.js:170:13)
    at async InverterDevice.handleReadingsEvent (/app/drivers/inverter/device.js:120:13)
```

There are no other errors in the log, but it appears that you have a flow that iterates without delay and checks power using a condition. There are only a few milliseconds between the checks. This could be the reason why the app hangs, obviously, only guessing. But I don't think Homey is capable of these endless loops without a delay. A better design to use the power changed trigger and use a logic condition to compare value of token, something like the picture below.

```
2025-07-05T05:13:11.016Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:11.016Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:11.016Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 201
2025-07-05T05:13:11.016Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:16.018Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:16.018Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 202
2025-07-05T05:13:16.018Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:16.018Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:16.018Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 202
2025-07-05T05:13:16.019Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:21.015Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:21.015Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 204
2025-07-05T05:13:21.015Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:21.015Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:21.015Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 204
2025-07-05T05:13:21.015Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:21.032Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:21.032Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 204
2025-07-05T05:13:21.032Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '100'
2025-07-05T05:13:21.032Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:21.032Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 204
2025-07-05T05:13:21.032Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '100'
2025-07-05T05:13:26.013Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:26.013Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 204
2025-07-05T05:13:26.013Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:26.014Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:26.014Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 204
2025-07-05T05:13:26.014Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
2025-07-05T05:13:31.013Z [log] [ManagerDrivers] [Driver:inverter] [SB 6.0-1] Condition 'power_condition' triggered
2025-07-05T05:13:31.013Z [log] [ManagerDrivers] [Driver:inverter] - inverter.power: 205
2025-07-05T05:13:31.013Z [log] [ManagerDrivers] [Driver:inverter] - parameter power: '300'
```

[image]

tjarb, 2025-07-05:
> is the card triggering getting the (in homey) stored device value, or does
> the card trigger a physical device read?
>
> As user I expect the first as there's a polling rate.
>
> yes the flows are triggered based on changes in the (SMA) gridpower meter.
> miliseconds seem to be very fast for the sma grid meters.

in addition to previous response:

the flow cards for getting (power) data are used in multiple flows. We do
not have control when these flows are triggered. Its always possible that
two flows trigger at the "same" time as this is per definition an event
based environment.

ricott, 2025-07-05:
> the card trigger a physical device read?
[capability read](https://github.com/ricott/sma.modbus/blob/e413c817f90eb7dd63f8da8adb4ecc46b72eaeb4/drivers/inverter/driver.js#L65)

There are 2000 installs of the app, and I'm trying to figure out what problems you may have that no one else has.

Regarding refresh intervals in general, I often encounter the belief that seconds are crucial. In both this SMA app and the Victron app I also manage, this is not accurate—vendors like Tibber, for example, control chargers at several-minute intervals. With the transition this autumn to 15-minute intervals instead of hourly timing, this may become somewhat more important.

tjarb, 2025-07-05:
im not sure how far the log went in the past, it ocured yesterday evening.
I used the PV production as light sensor and noticed the lights weren't
turned on.

oh, I meant did a flowcard triggering a device read, triggering the modbus
reques? or does it return a value from Homey memory. I really have not
taken a look into the code so thats just a guess.

I'm used to C/C++ in hard realtime envirronments ( sub-microsecond). I
really hate these higher level of coding.

ricott, 2025-07-09:
Please send a diag report. Have to be done before a restart of app. After restart the log is emptied

ricott, 2025-07-10:
Removal of batteries internal was never in a release, it is a mandatory section if you use the `measure_battery` capability. Addition of `meterPowerExportedCapability` shouldnt make a difference since it is the default value. 
What could trigger a freeze is an app install, happens quite frequently that apps hang after an upgrade to a new release.

raajschellekens-rgb, 2025-08-04:
Same problem here, occult every few days.
05534e26-8ddf-4b9a-b3fd-f177cc9916bd
Use to work flawless.

closed 2025-08-11 by ricott